SmokeMonsterPacks/Mega-Sg-Jailbreak #61: CJ Elephant Fugitive (Game Gear) not working at all.

DrBallistyx, 2019-10-25:
**Describe the bug**
Upon loading, the game will either display a black screen and/or garbled graphics.

**To Reproduce**
Simply load the ROM

**Firmware version**
Firmware v7.6

**ROM name and hash values**
CJ Elephant Fugitive (USA, Europe).gg (Taken from the No-Intro pack)  MD5: 30D140CBA0604578382CFF96ACA09F78

Brianvgplayer, 2019-10-26:
Isn't that a codemasters game? Have you tried changing the extension to .gcm?

DrBallistyx, 2019-10-26:
> Isn't that a codemasters game? Have you tried changing the extension to .gcm?

That seems to have done the trick. But now its stuck at "push start" in the main menu and no buttons on the controller work.

BooBerry, 2019-11-19:
This should be fixed in the new jailbreak firmware released today. :)

DrBallistyx, 2019-11-19:
I have tested it on Firmware v7.7 and can confirm it now works correctly :D

closed 2019-11-19 by DrBallistyx

DragonPokeBall, 2020-08-18:
> I have tested it on Firmware v7.7 and can confirm it now works correctly :D

You sure? Cause I have that current Firmware, and that game still won't work. >:P