SmokeMonsterPacks/Mega-Sg-Jailbreak #63: Sonic CD and Final Fight CD not working anymore on JPN model 1 Mega CD

alexcom84, 2019-11-20:
**Describe the bug**
Sonic CD it will load the background without any character sprites and the music will keep playing.
Final Fight CD will freeze on a black background after the initial Capcom logo.

**To Reproduce**
Steps to reproduce the behavior:
Start a new game on Sonic CD. Both NA and JPN versions have this issue.

**Firmware version**
Happens on both FW 4.7 and JB 7.7.
Downgrading to FW 4.6 will fix the problem.

Brianvgplayer, 2019-11-21:
I tried Sonic CD US and Final Fight CD JP and had the same issues. Sometimes Sonic CD locks up right away and sometimes it starts to play with a glitchy ring counter and occasionally blinking screen before locking up after time travel. I used multiple US bios for Sonic CD and the JP Mega CD2 bios for Final Fight.

scottjco, 2019-12-29:
I can confirm this issue with the PAL version of Sonic CD on my Mega CD 2.

Only its worse with mine, the level won't even load completely. Sonics sprite never loads and then the whole system resets. 

dropping down to 4.6 fixed it

the88g, 2020-04-28:
Unfortunately I'm also having this issue. Originally I had thought it was because I was trying to use a JP Mega CD BIOS with a European Mega CD 2 (as I've owned a JP copy of Sonic CD for years and never had the opportunity to play it). I've got a JP Mega CD 1 coming shortly, hoping it would be the fix and so I will try that, but something tells me I will have to downgrade to play - thanks for posting!

snicholls-orm, 2020-05-05:
Same issues, was there any fix? final fight and sonic cd

scottjco, 2020-05-05:
The only acceptable fix for this issue is using firmware 4.5, they haven't updated beyond 4.7 which is the issue. and 4.6 has much bigger problems at least for me.

I emailed them about sonic cd but I think since theres a PC port they don't care about that, more people need to email the devs over at analogue.

Brianvgplayer, 2020-05-06:
I'm not sure if the issues are just with Sonic CD and Final Fight, but it's disappointing if it's not a priority. Sega CD has had issues since earlier firmware and older firmware has various other issues that keep me from going back. It doesn't help that 4.7 and 7.7 are the only firmware that supports the new GG, Mark III, and SMS card adapters.

scottjco, 2020-05-06:
if they aren't going to fix it we can just use the jailbreak to play those. Its just annoying, since the Mega SG claims to be accurate to real hardware and it clearly isn't in these cases.

snicholls-orm, 2020-05-06:
I had an email from Analog and they said "Our developers are aware of this issue. Be sure to follow our firmware updates for a resolution to the problem. "

Jayde6, 2020-05-06:
Sadly this goes beyond just the Mega Sg, there are games that don't work correctly on the nt-mini and the super nt as well. Official firmware/jail broken it doesn't matter. I emailed support back in June of last year and got the same type of email, thanks for the info be sure to follow our firmware updates.

Very frustrating considering they keep bullet pointing complete compatibility and going on about how it's better then emulation. Ironically the SNES game that I found that doesn't work correctly didn't work right in emulators at first either, but most of the emulators have it fixed now. Analogue even used the game as one of their twitter game posts, they didn't respond when I commented on tweet about it not working.

scottjco, 2020-05-06:
on 4.6 you might notice some visual corruption or pixellation snicholls. Almost every game was full of them. 

And yeah, I don't buy the FPGA != Emulation bit myself, it IS emulation. Its just done in a different way, its just funny how silly people get over justifying spending a lot of money on one of these boards. There are certainly advantages but emulators and the analogues products have considerably more in common than people would like to accept.

snicholls-orm, 2020-05-15:
It is a shame that it isn't going to be fixed any time soon - I ended up getting a MegaSD to play the CD games.  Will keep an eye out for any updates.

the88g, 2020-05-15:
I have now logged a ticket with the support team and reiterated that bouncing between 3 firmwares just to play various games is terrible. I want to say, its not fit for purpose but I do love listening to music on it - YouTube rips of Mega Drive music is not the same anymore.

Brianvgplayer, 2020-05-28:
I could just play GG and sg1000 on Jailbreak, but I bought the adapters to use the actual carts, especially mark iii/sg-1000. I have no plans to roll back the firmware because of this. 

Sega cd always had some issue before 4.7/7.7 and the firmware loading still hasn't been 100% sorted out. The bug with third party ram carts for sega cd has been around for awhile and internal memory doesn't cut it for some games.

BooBerry, 2021-02-24:
This should be re-tested as soon as possible with the latest firmware that was released two days ago.

If it still doesn't work, there's new CD timing sliders which can be adjusted to see if it allows it to work. If you do have to adjust these for the Japanese Mega CD and you find a working timing it might be a good idea to post what it is here for others to know (and maybe the info can be passed along to kevtris so he can potentially make adjustments in the future). :)

closed 2023-06-25 by tjanas