SmokeMonsterPacks/Mega-Sg-Jailbreak #64: [Master System] [JB 7.7] Master System scroll shows tearing with all buffering modes

vanfanel, 2019-12-21:
Hi,

I have noticed that the Master System core shows strange scrolling hiccups and tearing in every buffering mode. 

The original hardware tends to struggle with scroll in 1-pixel scolling games, but I am not talking about normal Master System hiccups. The tearing and hiccups I am describing are not present are not present in real HW.

Showcases:
-In "Batman Returns", just walk right and left to force scroll movement. You can eliminate every enemy nearby if so you wish. Then look at the upper border of the screen for tearing.
-In "Gauntlet", move left and right (after killing every monster nearby if you so desire), and you will see small hiccups in the movement.
-In "Ys: Vanished omens" just move left and right around the initial town to see tearing on the upper part of the screen.

I repeat: These are NOT present in real hardware.

Thanks!

bumfacekilla, 2019-12-22:
I've tried to reproduce the issue using the games mentioned, but everything seemed fine for me and scrolling was smooth. The only time I have had issues with horizontal scrolling is if the system region and video output are mismatched (running a PAL system at 60Hz or NTSC at 50Hz).

vanfanel, 2019-12-23:
@bumfacekilla Have you tried 1080p mode? Its with that mode that those are more noticeable. 720p and 480p seem to be more smooth (not totally smooth either).
These tearing defects on the upper parts of the screen are definitively there and VERY noticeable. Tried different TVs.

bumfacekilla, 2019-12-25:
Yeah, I tried all resolutions and in all buffer modes. Aside from the stuttering caused by region/output Hz mismatches, the only other output issue I've seen is the occasional tearing when using single buffer mode (but that's to be expected).

another-James, 2020-05-16:
I’m getting the same issue. Visible on Bart vs the Space Mutants and Deep Duck trouble. My region and output frequency are matched. Using genuine carts on 4.7 firmware.

vanfanel, 2020-05-16:
The issue exists for many of us then.
I doubt we will see an updated fw soon, anyway...

G-Spindash, 2021-03-11:
Issue is still present on 7.8

Ulisses-Medeiros, 2021-03-26:
the same happens in Dragon Crystal (Europe, Brazil), Kesseiden (US ir JAP), Sonic (US, Europe), Castle of Illusion (US,Europe, Brazil) on JB 7.7 or JB 7.8,

GreatHierophant, 2021-06-07:
The tearing issue happens when the vertical height is set to 1080p/5x or some resolution near it.  Setting the vertical scaling to 4.5x or lower, or using a lower resolution, should avoid the tearing issue.  Not a perfect workaround I know, but the issue does not occur in the Nt Mini Noir, which is closely related to the Mega Sg in terms of software and hardware, so there is hope.

vanfanel, 2021-06-10:
> The tearing issue happens when the vertical height is set to 1080p/5x or some resolution near it. Setting the vertical scaling to 4.5x or lower, or using a lower resolution, should avoid the tearing [...]

It's just more obvious in 1080p/5x, but it happens in every video mode.
You can set 480p, for example, and it happens too, but it happens so fast that it looks like a single frame desynchronization: move left and right in any of the example games (in Batman Returns it is easy to do) without any sprites on screen, and you will see a "missing frame" in a while: it's the same tearing, just happening faster.

G-Spindash, 2021-06-12:
Can confirm. I usually run @ 720P  & it still happens.

ArkhangelAzrael, 2022-08-31:
I could perceive this problem quite easily on Aladdin with the default Fully Buffered and Screen Size 5x Height config, but I have noticed that I don't have any screen tearing with Fully Buffered mode if I use the config provided by Firebrandx here: https://drive.google.com/file/d/1jKV9oF7gW2PMf7Slh9iDv8gGhZzvUqXZ/view

Specifically, the 4.9x mode:

```
-- FirebrandX 4.96(3)x (fullscreen):
* set horizontal pos=65, vertical pos=32
320: 1640x1157 enable BOTH H and V (uncheck boxes)
256: 1497x1157 enable BOTH H and V (uncheck boxes)

```
I am using 1080p mode as well.

Scanlines don't work very well on that mode, but I have made them work good enough using:

```
Genesis Core Scanlines: 0 | 37 | 100 - Normal

```

tjanas, 2023-06-25:
@vanfanel Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

vanfanel, 2023-06-25:
This has been tested with latest JB 7.9 FW, and apparently it's fixed (at long last...)
Thanks all involved.

closed 2023-06-25 by vanfanel