SmokeMonsterPacks/Mega-Sg-Jailbreak #69: Steel Empire Graphic Errors on Stage 3

barbaro00, 2020-02-07:
Hi,

I have load the Steel Empire USA rom and always have image errors on the scroll, you can see the picture:

[image]

Thank you for your work is awesome!!

Steven-Pendleton, 2020-02-24:
Tested with both Japanese and US roms both from the Mega Sg cores menu and also on MegaSD. This never appeared on any of the 4 tests. I played it legit the first time, and then used the level select cheat to get there the other 3 times. I am also pretty sure this is stage 4, not 3.

frederic-mahe, 2020-02-29:
I am going to close that issue as it could be due to a bad ROM. Feel free to re-open the issue if the bug can be reproduced with a valid ROM or an original cartridge.

closed 2020-02-29 by frederic-mahe