SmokeMonsterPacks/Mega-Sg-Jailbreak #70: Desert Strike for Sega Master System locks up

Atariboy1982, 2020-02-10:
Hitting the Master System's pause button (i.e., the start button on the Genesis controller) to bring up your map/status screen in Desert Strike locks up the Mega Sg with the latest firmware.

Since it's a PAL exclusive, I've also tried setting region to Europe and utilizing various European bios files, which makes no difference. Nor does forcing 3 button mode. Issue was originally discovered with the system set to North America and a North American bios file. 

To reproduce, load the Desert Strike rom or the original cartridge, take off from the destroyer's deck, and fly towards the coast line. A handful of seconds in, press the Genesis controller's start button and the game will lock up and annoying static audio will play through your television's speakers. 

If you hit start very quickly after taking off, the map screen properly appears but the issue still happens upon attempting to exit the map screen.

A helpful AtariAge user with the original cartridge in his collection is getting identical behavior. Linked is his post, so it wouldn't appear to be an issue with a bad rom dump.

https://atariage.com/forums/topic/242970-fpga-based-videogame-system/?do=findComment&comment=4454682

Rom information:
No-Intro Name: Desert Strike (Europe) (En,Fr,De,Es)
(No-Intro version  20130731-235903)
ROM/File SHA-1: E6181BAEF80ECC88B3EB82A46CF93793F06E01F1

tjanas, 2023-05-06:
[Fixed](https://forums.atariage.com/topic/242970-fpga-based-videogame-system/?do=findComment&comment=5250700).

closed 2023-05-06 by tjanas

Atariboy1982, 2023-05-19:
It's not fixed and very much remains broken with the most recent jailbreak
firmware and thus shouldn't have been closed as fixed.

It's only fixed with the official non-jailbreak firmware with no signs that
the jailbreak firmware will ever be updated with the same fixes. The
jailbreak updates in the past always came about within a day or so after
the release of an official firmware update. Yet we're now about two weeks
out since the official update, signaling that the situation is somewhat
different than it was before.

While hopefully it happens eventually, only time will tell and we can no
longer just automatically assume it will eventually happen. Not after such
a big deviation from past practices. So I recommend holding off on the
closing of issues that have been fixed until the jailbreak counterparts of
each firmware actually receive said fixes. After all, it's the jailbreak
firmware that's the actual subject of the issues section of the
SmokeMonsterPack GitHub pages for the Super Nt/Mega Sg/Nt Mini Noir
(Although the issue reporting obviously also has directly benefited the
official firmware for each system).

-Leo Ames