SmokeMonsterPacks/Mega-Sg-Jailbreak #71: Set HDMI content type to 'game'

skooterblog, 2020-02-11:
HDMI specification allows for sources to specify the content type to be sent to the display, so it can automatically adjust viewing modes. Playstation 4 and other consoles set the content type to 'game', so the TVs in 'auto mode' can automatically enter the low latency 'game mode'.

On the other hand, Mega Sg doesn't set content type, so the 'game mode' has to be manually selected. It would be nice if Analogue Mega Sg could send the content type too.

mackoj, 2020-04-21:
Another great feature present since Xbox one is [VRR(Variable Refresh Rate)](https://www.hdmi.org/spec21Sub/VariableRefreshRate) it's available in [some TV](https://www.rtings.com/tv/tests/motion/variable-refresh-rate) from Samsung and LG pre-HDMI 2.1 and all TV compatible with HDMI 2.1.

mackoj, 2020-04-21:
Both of these feature has been integrated [since Xbox one](https://www.windowscentral.com/how-enable-xbox-one-auto-low-latency-mode) and both of those feature aim to reduce latency.
I would really love to see them in my Mega SG. Maybe we should ask **Analogue** the support to add them ?

SmokeMonsterPacks, 2021-02-24:
This didn't show up in the latest update so I'm closing the request to focus attention on bugs.

closed 2021-02-24 by SmokeMonsterPacks

mackoj, 2022-04-28:
😢