SmokeMonsterPacks/Mega-Sg-Jailbreak #77: Tiny Toon Adventures has no title screen music

therealinfidelity, 2020-05-25:
Using the latest firmware available as of 5-24-20, using the latest firmware on my Mega Everdrive X7, when loading the USA version of Tiny Tune Adventures, the title screen music does not play. It is not an issue with my rom or my Everdrive, since the cartridge & rom play the title screen music on my VA6 model 1 Genesis, and my VA2 model 2 Genesis.

tjanas, 2020-05-25:
I have confirmed this bug exists with loading the no-intro rom via the latest jailbreak firmware.

shroom2k, 2020-06-13:
I can confirm that I also have the same bug.

BooBerry, 2021-02-22:
With the new official v4.,8 firmware release, I can confirm this issue still exists.

P.S. It probably would've been helpful to clarify that it's **Tiny Toon Adventures - Buster's Hidden Treasure (USA)** with the broken title screen music as the other Tiny Toon game **Tiny Toon Adventures - ACME All-Stars (USA, Korea)** works fine.

therealinfidelity, 2021-02-22:
> With the new official v4.,8 firmware release, I can confirm this issue still exists.
>
> P.S. It probably would've been helpful to clarify that it's **Tiny Toon Adventures - Buster's Hidden Treasure [...]

Sorry. :-/

BooBerry, 2021-02-24:
It's quite alright, and I did ask kevtris about this in the Classic Gaming Discord channel and he's aware of the issue and looked into it but couldn't figure out why it happens yet. So it's on his known issues radar and hopefully he can figure it out when he gets the time to work on it again. :)

tjanas, 2021-02-24:
Apparently the issue is not present on the Genesis core for Mister, maybe some insight could be gained it?

AydanWatkins, 2021-02-25:
The game DOES have title screen music for me, I played it using an original cartridge using firmware v4.8 from Analogue. I know this is the Jailbreak firmware GitHub but surely if it works on official it should work on the Jailbreak firmware when updated too right?

BooBerry, 2021-02-25:
And you're playing `Tiny Toon Adventures - Buster's Hidden Treasure (USA)` right? `Tiny Toon Adventures - ACME All-Stars (USA, Korea)` works fine with the title screen music fortunately.

AydanWatkins, 2021-02-25:
Mine was Tiny Toon Adventures - Buster's Hidden Treasure (PAL) using an original official Megadrive cartridge. Surely it can't be down to Region??? If it plays on PAL it should on USA right? o_O

BooBerry, 2021-02-25:
Yes, it's a region difference. The EU PAL version works with music in the title screen, the US NTSC version does not.

PizzaCube, 2021-05-06:
it does not play on a real cart either.

tjanas, 2023-05-05:
@therealinfidelity @BooBerry @shroom2k @PizzaCube can you please re-test with the new official firmware update? https://www.analogue.co/support/mega-sg/firmware/4.9

BooBerry, 2023-05-06:
Yes, this issue is now fixed.

closed 2023-05-06 by tjanas