SmokeMonsterPacks/Mega-Sg-Jailbreak #83: Master System games don’t run with a Mega Everdrive Pro

dbbgba, 2020-08-03:
When launching Master System games from the Mega Everdrive Pro they just give me a black screen.
I installed the official 4.7 firmware and everything works correctly. I suspect the jailbroken firmware doesn’t reset the console properly.

cornellouis, 2021-01-20:
I am running jb 7.7 and mega everdrive pro 4.08 and SMS works when launched from the cart for me...

Infinitum8t, 2021-02-04:
> When launching Master System games from the Mega Everdrive Pro they just give me a black screen.
> I installed the official 4.7 firmware and everything works correctly. I suspect the jailbroken firmware doesn’t reset the console properly.

Did you find a solution to your problem. I'm having the same issue.

closed 2021-02-24 by SmokeMonsterPacks