SmokeMonsterPacks/Mega-Sg-Jailbreak #86: PAPRIUM of the MegaDrive does not work

Arcangels79, 2020-12-23:
The idiot of the creator of the game has not supported this game to work on our consoles, but he did have time to put in a vaporware game and laugh at all of us. I would like that those who are in charge of updating the firmware could put the support to this game (to play in cartridge mode, of course). Since those of us who have this game have suffered a lot to get it and because of the idiot of its creator we are unable to play it on this our beloved console, thank you.

El idiota del creador del juego no ha dado soporte para que funcione este juego en nuestras consolas, pero si le a dado tiempo para meter un juego vaporware y reirse de todos nosotros. Me gustaria que los que se encargan de actualizar el firmware pudieran meter el soporte a este juego (para jugar en modo cartucho,claro). Ya que los que tenemos este juego hemos sufrido mucho para conseguirlo y por culpa del idiota de su creador nos quedemos sin poder jugarlo en esta nuestra querida consola, gracias.

maniac79, 2020-12-23:
While I agree with the core message of this Issue, Issues are best submitted without emotions.
I too would be more than happy to have Paprium working on my Mega Sg, however, I'm not sure if this is something for the JB to fix or if it will have to be fixed on FW core level.

Arcangels79, 2020-12-24:
> 
> 
> While I agree with the core message of this Issue, Issues are best submitted without emotions.
> I too would be more than happy to have Paprium working on my Mega Sg, however, I'm not sure if this is something for the JB to fix or if it will have to be fixed on FW core level.

The ways are not appropriate, but when you are fed up with being pissed on ...
Anyway, do you know if the one that is in charge of the official firmware (Kevtris I think it's called) and the one of the jailbreak firmware will bring out compatibility with this cartridge?

SyntaxErrol, 2021-02-06:
Kevtris has identified and solved the issues. We are waiting for a new FW release.

Arcangels79, 2021-02-06:
> 
> 
> Kevtris has identified and solved the issues. We are waiting for a new FW release.

Thanks for the info!! ;)

closed 2021-02-24 by SmokeMonsterPacks