SmokeMonsterPacks/Mega-Sg-Jailbreak #88: Orignal speed mode

mistepien, 2021-01-27:
Now all you got is "Zero delay" forced to 50/60hz or some buffered modes but with lag.
It would be great to have another option, namely to have original speed without any buffers -- in Mega Everdrive PRO there is fpga support for NES but it needs original speed console. Why force people to have some buffered and laggy mode?

SegaSnatcher, 2021-02-03:
The issue is Analogue consoles are HDMI certified products and I don't think you can sell said console with out of spec HDMI timings, so if this were to be added it would definitely need to be a JB exclusive feature.

SmokeMonsterPacks, 2021-02-24:
This didn't show up in the latest update so I'm closing the request to focus attention on bugs.

closed 2021-02-24 by SmokeMonsterPacks