SmokeMonsterPacks/Mega-Sg-Jailbreak #89: reset instruction bug

krikzz, 2021-02-06:
After reset instruction execution system start data fetching from cart before than reset signal (vres B27) actually released. It may result crashes if console used in pair with reset sensitive devices (flashcarts, for example).
Some rare games may using this instruction, here is few of them: 
Risk, Paperboy 2, Boxing legends of the ring.

[image]

tjanas, 2023-05-05:
@krikzz please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

closed 2023-06-25 by tjanas

tjanas, 2023-06-25:
Per JB7.9 changelog:
Fixed RESET instruction's output so reset is only asserted during the instruction.