SmokeMonsterPacks/Mega-Sg-Jailbreak #90: bus timings bug

krikzz, 2021-02-06:
In specific conditions bus read cycle may be shorter than usually. This results Virtua Racing crashes on EverDrive when select "Free Run". Seems it sometimes happen after write cycles to the VDP data port (0xC00000) in pair with dbra instruction followed. In buggy read cycle, timings between as, cas0, ce_0, asel looks differently and  cycle is shorter than usually.  In case of buggy cycle console fetch data before than valid value (0xFFF8) placed on the bus.

[image]
[image]

xtcrefugee, 2021-02-06:
Free Run appears to work fine in VR on a Mega SD. Maybe it's a Mega EverDrive PRO bug?

krikzz, 2021-02-06:
> Free Run appears to work fine in VR on a Mega SD. Maybe it's a Mega EverDrive PRO bug?

I made faster rom access and now it works on mega ed also, but on original console it worked fine without that fix. This behavior isn’t exist on original hardware. It may affect some other games, especially originals carts which often had pretty slow rom chips.

tjanas, 2023-05-05:
@krikzz please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

closed 2023-06-25 by tjanas

tjanas, 2023-06-25:
Per JB7.9 changelog:
Fixed bus timing issue after certain writes - this affected Virtua Racing in some situations.