SmokeMonsterPacks/Mega-Sg-Jailbreak #91: Paprium fails to boot into full game when Mega Sg is connected to display via HDMI switch

pvalkone, 2021-02-25:
**Describe the bug**
Paprium fails to boot into the full game when the Mega Sg is connected to the display via a HDMI switch. The game loads fine when the console is connected directly to the display.

**To Reproduce**

1. Connect Mega Sg to display via HDMI switch
2. Turn on the console
3. The Mega Sg boots into the Paprium mini-game
4. Reset Mega Sg
5. The Mega Sg still boots into the mini-game

--

1. Connect Mega Sg directly to display
2. Turn on the console
3. The Mega Sg boots into the full game

**Expected behavior**
The Mega Sg should boot into the full game on the second boot when the console is connected to the display via a HDMI switch.

**Screenshots**
N/A

**Firmware version**
JB7.8 (Feb 14 2021)

**Relevant Mega Sg settings**
All settings were first reset to default.
```
Settings > System > Startup options

Boot sequence delay: +0
Cartridge direct: [x]

Settings > System > Hardware

Japan (NTSC domestic): [x]
Auto region detection [x]
```

**ROM name and hash values**
Paprium Classic (JUE) cartridge (black).

**Additional context**
The AV equipment I'm using is as follows:

- Panasonic [VT60](https://shop.panasonic.com/support-only/TC-P55VT60.html) plasma TV
- [ATEN VS0801HB](https://www.aten.com/global/en/products/professional-audiovideo/video-switches/vs0801hb/) HDMI switch
- Generic HDMI cables

Others have also reported similar behaviour when using a HDMI switch in [the comments of this video](https://www.youtube.com/watch?v=MgEKjkaahlI&lc=UgwCpiatXCUFoWhbKKZ4AaABAg).

pvalkone, 2021-03-01:
Ditto for [AVRs](https://twitter.com/Needlecrash/status/1364780916929949698).

tjanas, 2023-05-05:
@pvalkone please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

pvalkone, 2023-05-05:
> @pvalkone please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

After updating to v4.9, the Mega Sg now boots to the full game (with the same settings and signal chain as mentioned in the initial report). Thanks for fixing this!

closed 2023-05-05 by tjanas