SmokeMonsterPacks/Mega-Sg-Jailbreak #94: 256 Display Aspect Ratio formula incorrect for MD core:

Firebrandx, 2021-03-11:
I forgot to post this bug report last time, so I'm getting it done now after the latest firmware (JB 7.8). The problem is the formula for DAR is off on the MD core for 256 mode. I had to measure the 'normally' active pixels in a paint program since the menu width settings are based on overscan. Below are 5x and 4x examples showing the currently incorrect labels, and the correct values they should be at to honor the correct DAR  formula:

Genesis 256: 

4:3 for 16:9 -  5x 1530 (wrong), 1554 (correct).
4:3 for 16:10 - 5x 1699 (wrong), 1725 (correct).

4:3 for 16:9 -  4x 1224 (wrong), 1243 (correct).
4:3 for 16:10 - 4x 1360 (wrong), 1381 (correct), remove the "5x256" part of the label as well.

tjanas, 2023-06-25:
@Firebrandx Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

Firebrandx, 2023-06-25:
I won't be able to do testing until next month as I'm moving into my new place. My apologies, but I will have them done in a few weeks with new videos! Cheers!

tjanas, 2023-06-25:
Good luck on the move!

Firebrandx, 2023-06-25:
Thanks! Shouldn't need much luck as it's literally right across the street. LOL!

closed 2023-06-26 by tjanas