SmokeMonsterPacks/Mega-Sg-Jailbreak #95: 32X Roms won't play off the Pro with latest firmware 4.8/7.8

bntruong0325, 2021-03-21:
**Describe the bug**
Using a 32X attachment and the Analogue DAC, did not have any issues playing 32X cartridge game ROMs off the Mega EverDrive Pro.  This changed with MegaSG firmware updates 4.8 & 7.8 jailbreak.  Now 32X cartridge game ROMs do not load completely when selected (resulting in either a blank black or blue screen) using the EverDrive Pro. 32X commercially made game cartridges still work fine, however.

**To Reproduce**
Steps to reproduce the behavior:
Use EverDrive Pro flashcart with latest flashcart firmware 4.11. "Run cartridge" then Try to load 32X cartridge game ROM from either using the Mega SG with the 32X attachment and connected through the DAC on the latest Mega SG firmware 4.8/7.8

**Expected behavior**
Most/all 32X cartridge ROMs load and are playable off of the EverDrive Pro as they were previously on firmware 4.7/7.7 for the MegaSG

**Firmware version**
4.8/7.8

**Relevant Mega Sg settings**
Used factory reset default settings for both 4.8 and 7.8 firmwares.

**Additional context**
Did not have this issue using EverDrive Pro for running 32X cartridge ROMs on firmwares 4.7/7.7 with 32X attachment and DAC previously. 32X commercial game cartridges still play fine.  When flashing "back" to previous firmwares 4.7/7.7 again, 32X cartridge ROMs did load and play properly off of the EverDrive Pro once more.  

I also have a TerraOnion MegaSD, but cannot get it to play 32X cartridge game ROMs consistently on 4.7/7.7 (sometimes ROMs will load normally, sometimes it will load and play with audio bugs, sometimes I just get a blank black or blue screen) although 32X cartridge ROMs do not work at all on 4.8/7.8.

scottlinux, 2022-01-11:
I can confirm the same behavior, fwiw.

tjanas, 2023-05-05:
@bntruong0325 @scottlinux please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

scottlinux, 2023-05-05:
Unfortunately no change in behavior, but perhaps worse. Attempting to use any retail 32X cart results in a black screen. A mega everdrive pro lets one select a game, but then any 32X title just goes to a black screen. I realize using a Mega SG + DAC and 32X unit is an edge case.

BooBerry, 2023-05-06:
What revision of the 32X do you have? Because if it's a VA0 32X you'll need the 32X Expansion Unit (aka the bus terminator) to play 32X games correctly on the Mega Sg. The bus terminator as-is won't fit on the Mega Sg's expansion slot, you have to remove the terminator from its plastic housing and plug it in the correct orientation (otherwise the system will bootloop and *could* cause damage if it's upside-down).

It's the same with the MegaSD as well (make sure you're using hard reset in its options with 32X games). The need for the bus terminator with VA0 32Xes is mentioned in Analogue's documentation for the 32X + DAC support.

scottlinux, 2023-05-10:
Ah thank you! Mine has Model MK-84000A and I see VA0 in the cart slot. Thank you so much, I will track down a bus terminator!

BooBerry, 2023-05-10:
Actually, MK-84000A usually means you have a VA1 32X with the bus termination fix built-in. You'd have to check the PCB of the board that plugs into the Genesis (or Mega Sg in this case) to look for VA1 whereas the PCB of the cartridge slot of the 32X will always be VA0.

scottlinux, 2023-05-13:
Ok thank you, my 32X unit does say VA1 on the PCB of the Genesis cart slot. After some unplug-replug and verifying connections, I am able to play 32X retail carts with firmware 4.9!

However, the Mega Everdrive Pro does not appear to load, even though I see it has power (flashing light on the cart at power on). I was able to get to the Mega Everdrive Pro menu once but I'm not sure if that was an anomaly.

closed 2023-06-25 by tjanas