SmokeMonsterPacks/Mega-Sg-Jailbreak #96: XE-1AP analog mode does not work

Steven-Pendleton, 2021-04-01:
(do not submit bugs if you are using any means of powering your Mega Sg other than the OEM power supply. Only use the original PSU and USB cable when bug testing)

**Describe the bug**
XE-1AP does not work at all in analog mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure XE-1AP for MD analog mode
2. Press buttons or move stick
3. Nothing happens at all in any situation
4. With After Burner II cart, notice that the analog mode indicator is not displayed on title screen

**Expected behavior**
Buttons and stick should work in menus. Should provide full analog control in games like After Burner II and Musha Aleste. After Burner II should automatically detect XE-1AP in analog mode and indicate this on the title screen and also provide special XE-1AP configuration menu in options.

**Firmware version**
7.8

**Relevant Mega Sg settings**
N/A

**ROM name and hash values**
N/A

**Additional context**
Digital mode seems to work fine. Navigate to Mega Sg controller test function in digital mode and running the test in digital mode works fine. Opening the controller test menu in digital mode and changing from digital to analog will show a continuous press of the B button.

GreatHierophant, 2021-05-24:
Did you try Controller Passthrough mode?  That is intended to allow unusual peripherals to communicate directly with the Mega Sg's Genesis Core without any kind of input processing.

Steven-Pendleton, 2021-05-24:
Yes. No change.

Steven-Pendleton, 2023-05-05:
Too lazy to do it now. I'll do it eventually.

closed 2023-06-25 by tjanas

tjanas, 2023-06-25:
Per JB7.9 changelog:
Added support for XE-1AP analog controller.