SmokeMonsterPacks/Mega-Sg-Jailbreak #97: Virtua Racing Original Cartridge Doesn't Work - Black Screen.

ajsliter, 2021-05-20:
Description
I recently got a Virtua Racing cart after Analogue confirmed the Virtua Racing Cartridge works on the Mega SG and there had been multiple examples of video evidence that it is true. I'm running the latest jailbroken firmware. I've verified the SVP chip exists and it is an original SEGA Cart. I've verified the capacitors are in good shape. I have also verified with the seller that the cartridge was tested and works with a Model 2 Sega Genesis prior to shipping to me. I have no original hardware to test the cartridge. When I plug in the cartridge to my Mega SG and power on and after pressing run cartridge I get a black screen. No amount of waiting will change that outcome (I've waited for it to boot for about 10 minutes maximum before trying to reset or hard reboot my Mega SG). I've tried official firmware 4.8 and the latest jailbroken firmware to no avail.

**To Reproduce**
Steps to reproduce the behavior:
1. Put in Virtua Racing cart to cartridge slot of Mega SG
2. Turn on Mega SG
3. Select Run Cart
4. Press A
5. Black Screen

**Expected behavior**
The cartridge should boot up and play.

**Firmware version**
Latest 7.8 Version Jailbreak / v4.8 Official.

Screenshots:
[image]

Steven-Pendleton, 2021-05-20:
The SVP chip itself can fail and cause this. Mine works fine on the Mega Sg, so it is possible that your SVP chip has failed.

xnopasaranx, 2022-03-12:
not sure if this helps anyone, but I thought my virtua racing was not working either, until I switched the region to the correct one. it seems auto detection failed for my cartridge.