SmokeMonsterPacks/Mega-Sg-Jailbreak #98: Demons of Asteborg rom crashes

retiras, 2021-08-05:
**Describe the bug**
Demons of Asteborg (World) (Aftermarket) (Unl).md rom crashes in the first stage.

**Firmware version**
Latest 7.8 Version Jailbreak / v4.8 Official.

**ROM name and hash values**
Demons of Asteborg (World) (Aftermarket) (Unl).md
CRC32:
753bfc8f
MD5:
1abb8fc4e67b6e10d7902f2c8f3be7e4
SHA-1:
2c59f4fa40a5e4da012f8fb3aa4970a8f462f1f4

Steven-Pendleton, 2021-08-16:
Where did you get this ROM? This looks like some pirate version, judging by that (World) (Aftermarket) (Unl) stuff. Mine is from the Kickstarter, emailed directly to me by Neofid, and ROM Hasher gives me these values:

ROM CRC32    67283FBE
ROM SHA-1    98A143B2312A062D640952700CD0D2DB78326FE2

Regardless, I get something different on 7.8 when run from the SD card slot: garbled intro, then a functional menu, then rather than crashing when I start the game, I get a garbled screen and I fall to my death repeatedly.

retiras, 2021-08-16:
> Where did you get this ROM? This looks like some pirate version, judging by that (World) (Aftermarket) (Unl) stuff. Mine is from the Kickstarter, emailed directly to me by Neofid, and ROM Hasher [...]

It´s from the last no-intro romset and I have the same issue. You explained it better.

ShinKaz, 2021-08-23:
Same issue here, rom from KS too.

itsTrifle, 2021-09-05:
FYI I get similar issues with the v1.1 ROM (from Steam installation)
* Neofid logo animation gfx & sound garbled
* Attract sequence freezes during a fade to white 
* Black screen when starting game 

CRC32 AACB216E

jtorrad, 2021-09-06:
I also experienced the same issues with 1.1 ROM (I bought a physical cartridge) on 7.8 jailbreak.
SHA1 7525D9F54667B2E26FEAEA265F16316E2C7D96EC

issue description:
* gabled graphics and sound from logo animation
* functional start menu
* garbled graphics with the continuously character falling

ProjectLittleMan, 2022-01-05:
> Where did you get this ROM? This looks like some pirate version, judging by that (World) (Aftermarket) (Unl) stuff. Mine is from the Kickstarter, emailed directly to me by Neofid, and ROM Hasher [...]

We learned today that this specific ROM was leaked by the DoA team as an anti piracy mechanism. 
[image]

tjanas, 2023-05-05:
@jtorrad can you test your physical cartridge with https://www.analogue.co/support/mega-sg/firmware/4.9

BooBerry, 2023-05-06:
I tested the latest ROM of Demons of Asteborg (v.1.11) on both the MegaSD and Mega Everdrive Pro and the game works fine on both on the Mega Sg with the 4.9 firmware.

closed 2023-05-06 by tjanas

L10N37, 2023-05-06:
@BooBerry 

I only purchased the ROM and use a Jail Broken FW :( i've owned it since release and never being able to try it.

tjanas, 2023-06-25:
@L10N37 please try JB7.9 and if that does not resolve the issue, please open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB