SmokeMonsterPacks/Mega-Sg-Jailbreak #101: MegaSD does not load 32X roms on JB 7.8

RevQuixo, 2021-08-25:
Roms (with the exception of Spider-Man: Web of Fire) load on 4.8 using the MegaSD (hard reset in options).  They have worked in prior fw as well.

7.8 introduced a fix for Spider-Man: Web of Fire that causes  all 32X roms to black screen when using the MegaSD.  You can reset (Up + Start) and load cartridge again a couple of times and usually trick the MegaSD into loading the game but never on first boot.  Assuming whatever timing /reset change was made for Spider-Man broke everything else (including Spider-Man)

scottlinux, 2022-01-11:
Same behavior as #95

BooBerry, 2023-05-06:
If your 32X is a revision VA0, you'll need the 32X Expansion Unit (aka the bus terminator) connected to the expansion port when using it with a Mega Sg.

More detail about that here: https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/issues/95#issuecomment-1537141751

tjanas, 2023-05-07:
@RevQuixo what is the behavior with the latest official firmware?

tjanas, 2023-06-26:
Closing issue here since JB7.8 is no longer supported and should be tested with JB7.9, and tracked in the new repo.

closed 2023-06-26 by tjanas