SmokeMonsterPacks/Mega-Sg-Jailbreak #103: Mega Sg Jailbreak v7.8

lilo-san, 2021-10-03:
**Describe the bug**
Screen Tearing at random, particularly visible on the top of the screen.
It only happens when combining: Fully Buffered and Screen Size 5x Height
Doesn't happen when combining: Fully Buffered and Screen Size 4x Height
Doesn't happen when combining: Zero Delay and Screen Size 5x Height

**To Reproduce**
Steps to reproduce the behavior:
1. Toggle setting '...'
Buffered Mode - Fully Buffered
Screen Size - 5x Height
2. Load ROM '....'
Sonic 3D Blast
3. Play through '....'
Go around the first level for 1-2 minutes and you should spot a couple of appearances of it. This game scrolls on all directions so is an easy way to reproduce it.
4. See error

**Expected behaviour**
I expect to not have random appearances of screen tearing.

vanfanel, 2022-01-23:
Possible the same issue as this:

https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/issues/64

..But no one seems to care about it. I gave the core author visual evidence a year ago.

PizzaCube, 2022-03-04:
i am getting this too but was not sure if it was normal for the game since it is so close to the crt overscan area and i just never saw it before.

ArkhangelAzrael, 2022-08-31:
I could perceive this problem quite easily on Aladdin with the default Fully Buffered and Screen Size 5x Height config, but I have noticed that I don't have any screen tearing with Fully Buffered mode if I use the config provided by Firebrandx here: https://drive.google.com/file/d/1jKV9oF7gW2PMf7Slh9iDv8gGhZzvUqXZ/view

Specifically, the 4.9x mode:

```
-- FirebrandX 4.96(3)x (fullscreen):
* set horizontal pos=65, vertical pos=32
320: 1640x1157 enable BOTH H and V (uncheck boxes)
256: 1497x1157 enable BOTH H and V (uncheck boxes)

```
I am using 1080p mode as well.

Scanlines don't work very well on that mode, but I have made them work good enough using:

```
Genesis Core Scanlines: 0 | 37 | 100 - Normal

```

hergel, 2023-01-21:
I have the same issue, as long as I have set to 5x height I can clearly see a screen tear on top of the screen, this goes away if I change to 4.5x height.

hergel, 2023-01-21:
My issues went away when i used FirebrandX SMS config:

Set horizontal pos=2, vertical pos=56 1462x1200 enable H, disable V

spoonmanx, 2023-05-07:
Looks like it was fixed with the new 4.9 Analogue firmware.
https://www.analogue.co/support/mega-sg/firmware/4.9
_(2023/05/04] "Fixed screen tearing on SMS"_

Now we just need a new JB version. 😉👍

closed 2023-06-25 by tjanas