SmokeMonsterPacks/Mega-Sg-Jailbreak #113: Columbus Circle 2019 cartridges Gleylancer and 16bit Rhythm Land no longer work under firmware 7.7/7.8

nihilisation, 2022-09-07:
Mega Drive Columbus Circle's 2019 cartridge releases "Gleylancer" and "16bit Rhythm Land" no longer work/boot under firmware 7.7 onward. They are working with firmware up to 7.6/4.6 .

Describe the bug:
Attempts of booting 2019 "GleyLancer" or "16bit Rhythm land" cartridges give a black screen

To Reproduce:
Insert cart, power on, run cart

Firmware version:
7.7 and 7.8, worked fine on 7.6 or "official" 4.6

tjanas, 2023-05-05:
Can you please re-test with the latest official firmware?

tjanas, 2023-05-05:
@nihilisation it was reported as fixed with https://www.analogue.co/support/mega-sg/firmware/4.9

nihilisation, 2023-05-07:
@tjanas I cannot test currently. I will update this post in a few days, around 12/05/2023. Thanks for the heads up, I would have missed the last update without your comment !

closed 2023-06-25 by tjanas

tjanas, 2023-06-25:
Per JB7.9 changelog:
Added support for Columbus Circle games (16 Bit Rhythm Land, etc).