SmokeMonsterPacks/Mega-Sg-Jailbreak #116: New firmware by Analogue, fixing bugs

hansvangent, 2023-05-09:
(do not submit bugs if you are using any means of powering your Mega Sg other than the OEM power supply. Only use the original PSU and USB cable when bug testing)

**Describe the bug**
Analogue released a new firmware that fixes some bugs a couple of days ago:

https://www.analogue.co/support/mega-sg/firmware/4.9

Are you planning on taking that firmware and using it to update the jailbreak?

**Firmware version**
4.9

eewokee, 2023-06-13:
Please do SmokeMonster and please do not let us in the dark about when to expect the updated jailbreak. Much appreciated!

tjanas, 2023-06-23:
SmokeMonster does not create jailbreak firmware.

tjanas, 2023-06-25:
@hansvangent please download JB7.9

closed 2023-06-25 by tjanas