SmokeMonsterPacks/Mega-Sg-Jailbreak #117: Mega Man The Sequel Wars Episode Red

mario-64, 2023-05-26:
The new Genesis homebrew game “Mega Man: The Sequel Wars - Episode Red” is completely unplayable on Mega SG due to severe audio issues. Tested on firmware 4.9. The game runs fine on MiSTer and real Genesis hardware. Dev had this to say regarding Mega SG: "The sound driver needs to be emulated pretty precisely. Some older emulators like Gens also suffer this issue."

https://woodfrog.itch.io/mega-man-the-sequel-wars-episode-red

Steps to reproduce the behavior:
1. Power on Mega SG
2. Load the game via flashcart (MED Pro or Mega SD)
3. You will hear numerous audio issues such as no capcom jingle when the logo is shown and screeching, squealing audio when a tune should be playing. Compare against real hardware or MiSTer to confirm. 

**Expected behavior**
Sounds and music should play as they do on real hardware and MiSTer

**Firmware version**
Official firmware v4.9

**ROM name and hash values**
Mega Man The Sequel Wars Episode Red.bin

mario-64, 2023-05-26:
How exactly do you want me to verify it?

mario-64, 2023-05-26:
I confirmed this is a regression in official firmware 4.9. The game plays correctly in 4.8

impetus-maximus, 2023-06-22:
> I confirmed this is a regression in official firmware 4.9. The game plays correctly in 4.8

I can personally confirm the Audio bug exists in 4.9 on my Mega Sg i just received. Audio plays fine on a MegaRetronHD/Mega Everdrive x5.

Others have stated the same
https://itch.io/post/7843085
https://forums.atariage.com/topic/242970-fpga-based-videogame-system/?do=findComment&comment=5273465

mario-64, 2023-06-23:
Analogue is aware and says it will be fixed in the next firmware. Problem is, based on their past we shouldn’t expect it until sometime next year 😞

tjanas, 2023-06-25:
@mario-64 Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

mario-64, 2023-06-25:
This issue is a regression in official firmware 4.9 which the new JB is based on. It has been confirmed the issue is present in the new JB as well. Analogue is aware and says the defect will be addressed in a future firmware. We will see if that holds true.

tjanas, 2023-06-25:
@mario-64 can you please duplicate this issue at https://github.com/analoguejb/Analogue-Mega-Sg-JB

closed 2023-06-25 by tjanas