SmokeMonsterPacks/Mega-Sg-Jailbreak #118: Unable to install JB7.8 or Downgrade to 4.8

camohiddendj, 2023-06-23:
(do not submit bugs if you are using any means of powering your Mega Sg other than the OEM power supply. Only use the original PSU and USB cable when bug testing)

**Describe the bug**
Unit does not flash. On boot, unit LED flashes blue once then turns white. 
Verified that the SD card is at least being read by the unit via loading in a Font via SD in the system menu

**To Reproduce**
Steps to reproduce the behavior:
Attempt to flash any firmware other than 4.9 on latest Mega SG production run.

**Expected behavior**
Firmware to flash

**Screenshots**
N/A

**Firmware version**
4.9

**Relevant Mega Sg settings**
Factory Default

**ROM name and hash values**
N/A

**Additional context**
Tried multiple SD cards. Tried formatting with rufus, and sd formatter. as well as tried mkdosfs, multiple times, including specifying -F 32 and -S 512 paramaters. 

I used DD to make an image of the SD card I was using to try and flash the system in case it helps you to verify that I did or did not format the SD card correctly
https://files.d2cp.win/msg_firmware_ver4.8.img.gz (2gb raw image, gzip to 11mb)
https://files.d2cp.win/msg_firmware_verJB7.8.img.gz

juraisa, 2023-06-23:
This shouldn't be an open issue, downgrading the system is not a jailbreak related issue. Please use the correct avenue, aka, contacting Analogue, not SmokeMonster who does not work for them.

tjanas, 2023-06-23:
> This shouldn't be an open issue, downgrading the system is not a jailbreak related issue. Please use the correct avenue, aka, contacting Analogue, not SmokeMonster who does not work for them.

Incorrect. This github also tracks issues related to official firmware, and this is an issue that impacts being able to install JB firmware.

SmiteOfHand, 2023-06-23:
Just wanted to confirm that I am not seeing this issue on an initial run 2019 model so should be able to rule out fw4.9 as being any sort of issue. [Same 32GB SD card](https://www.amazon.com/gp/product/B06XWMQ81P/) used for years now with no special formatting, directly out of the package.

https://i.imgur.com/6U4ookX.png
https://i.imgur.com/bfBQR5F.png

Firmware Checksums. Downloaded today directly from Analogue.co and Github. Loaded as the only firmware file on the root of the SD card.

msg_firmware_ver4.9.bin
MD5 : 7c8d173eec8e6746b0a7bcff38f7683f

msg_firmware_verJB7.8.bin
MD5 : 8d7ddd48ef951c7a4518cde82adf0412

camohiddendj, 2023-06-23:
[image]

Also confirmed as a new hardware revision by kevtris in FPGaming Discord

tjanas, 2023-06-24:
So it seems like we all need to wait for the upcoming jailbreak firmware to be in line with the latest official firmware, which is compatible with both hardware variants.

impetus-maximus, 2023-06-24:
> So it seems like we all need to wait for the upcoming jailbreak firmware to be in line with the latest official firmware, which is compatible with both hardware variants.

agreed

tjanas, 2023-06-25:
Fixed with JB7.9

closed 2023-06-25 by tjanas

tjanas, 2023-06-25:
@camohiddendj please re-test with JV7.9

camohiddendj, 2023-06-25:
Can confirm, this resolves #118 in regards to the jailbreak